GyroGearl00se/ha_froeling_lambdatronic_modbus #1: Question: Waveshare RS232/RS485 to Ethernet Converter vs. ESP32

krauskopf, 2025-06-13:
Hi,

thanks for the nice project. I have a question: Is there a reason that you recommend the **Waveshare RS232/RS485 to Ethernet Converter** and not the **ESP32** based modbus converter from https://github.com/GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200?
I assume, that this integration is compatible to both, but just want to be sure.

Thanks!
Sebastian

GyroGearl00se, 2025-06-13:
Hi @krauskopf ,

There is no recommendation from my side. I just liked the idea of being able to provide different options.

Overall I had both options running simultaneously, and both worked fine for me.

Some might just prefer to buy a TCP RS233 adapter as it is pretty much plug & play.
Others don't mind messing around with ESPHome, yaml files, flashing the device, 3D printing a case etc.

krauskopf, 2025-06-13:
Hi @GyroGearl00se,

thanks for the quick reply. Was just wondering, because you only mentioned the waveform explicitly. But good to know....

Thanks again!

krauskopf, 2025-06-13:
Proposal:

This integration has been tested with the **Waveshare RS232/RS485 to Ethernet Converter**, but other adapters should also work (e.g. the **ESP32 based RS232 to WiFi adapter** as described [here](https://github.com/GyroGearl00se/ESPHome-Froeling-Lambdatronic_3200)) or any other TCP Modbus adapter.